ModusCreateOrg/evade2 #337: Freeze after upload

MrBlinky, 2017-12-23:
After uploading Evade 2. Arduboy will lock up due to USB interrupts left enabled by the bootloader in combination with the undefined USB interrupt vector.

add **UDIEN = 0** before init() to fix this.
also adding **__attribute__ ((OS_main))** to main will save a few extra bytes

```
int __attribute__ ((OS_main)) main(void)
{
    UDIEN = 0; //disable possible enabled USB interrupts from bootloader
	init();

	initVariant();
	
	setup();
    
	for (;;) {
		loop();
		if (serialEventRun) serialEventRun();
	}
        
	return 0;
}
```

dxxb, 2017-12-23:
Thanks @MrBlinky. I had noticed the issue but did not investigate the reason.

jaygarcia, 2017-12-24:
Thank you @MrBlinky.  I noticed this as well but wasn't sure if it was my old & worn devices acting up. I tend to be hard on them while developing. :(

closed 2017-12-31 by jaygarcia