ModusCreateOrg/evade2 #338: Cannot build 1.21

dxxb, 2017-12-23:
If I checkout version 1.21 (current master HEAD) and run cal.sh I get

```
make: *** No rule to make target `sound/evade2_00_intro.h', needed by `build-leonardo/Sound.cpp.o'.  Stop.
```

possible cause is 5df3417fbf4aaf86c2c7c00c87e35a26f8f1be82. Anyone else is seeing this?

jaygarcia, 2017-12-24:
@dxxb Yes. There is an issue with pre-built object files. If you uncomment out the `make clean` command in cal.sh, the issue gets resolved.  IDK how else to resolve this as Sound.cpp doesn't require that file any more.

dxxb, 2017-12-24:
Weird, when I tried running make clean manually the issue didn’t go away. I think the easiest way to fix this would be to run make clean in cal.sh

jaygarcia, 2017-12-24:
`make clean` by itself won't work because the ARDUINO_DIR or whatever it is called is not passed to the make file.  The script does this for us easily :)

dxxb, 2017-12-24:
Yep, I saw that after I wrote my previous comment. I have uncommented ```make clean``` in _cal.sh_ and all is good.

closed 2017-12-31 by jaygarcia